pimoroni/skywriter-hat #2: smbus dependency not found

goliatone, 2015-01-25:
When you run `setup.py` the install fails as it can not find a package/link for `smbus`:

```
running install
Checking .pth file support in /usr/local/lib/python2.7/dist-packages/
/usr/bin/python -E -c pass
TEST PASSED: /usr/local/lib/python2.7/dist-packages/ appears to support .pth files
running bdist_egg
running egg_info
creating skywriter.egg-info
writing requirements to skywriter.egg-info/requires.txt
writing skywriter.egg-info/PKG-INFO
writing top-level names to skywriter.egg-info/top_level.txt
writing dependency_links to skywriter.egg-info/dependency_links.txt
writing manifest file 'skywriter.egg-info/SOURCES.txt'
reading manifest file 'skywriter.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'skywriter.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv6l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv6l-2.7
copying skywriter.py -> build/lib.linux-armv6l-2.7
creating build/bdist.linux-armv6l
creating build/bdist.linux-armv6l/egg
copying build/lib.linux-armv6l-2.7/skywriter.py -> build/bdist.linux-armv6l/egg
byte-compiling build/bdist.linux-armv6l/egg/skywriter.py to skywriter.pyc
creating build/bdist.linux-armv6l/egg/EGG-INFO
copying skywriter.egg-info/PKG-INFO -> build/bdist.linux-armv6l/egg/EGG-INFO
copying skywriter.egg-info/SOURCES.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying skywriter.egg-info/dependency_links.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying skywriter.egg-info/requires.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
copying skywriter.egg-info/top_level.txt -> build/bdist.linux-armv6l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/skywriter-0.0.2-py2.7.egg' and adding 'build/bdist.linux-armv6l/egg' to it
removing 'build/bdist.linux-armv6l/egg' (and everything under it)
Processing skywriter-0.0.2-py2.7.egg
creating /usr/local/lib/python2.7/dist-packages/skywriter-0.0.2-py2.7.egg
Extracting skywriter-0.0.2-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding skywriter 0.0.2 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/skywriter-0.0.2-py2.7.egg
Processing dependencies for skywriter==0.0.2
Searching for smbus
Reading http://pypi.python.org/simple/smbus/
Couldn't find index page for 'smbus' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for smbus
error: Could not find suitable distribution for Requirement.parse('smbus')
```

Gadgetoid, 2015-01-25:
This is because smbus isn't available in the Python package index, so Python is incapable of resolving it as a dependency. It's irritating :( However, the solution is pretty simple, it's available via apt. So try:

    sudo apt-get install python-smbus

And try again ;)

goliatone, 2015-01-26:
Ok, did execute the `apt-get` command, and `smbus` is now showing up when `pip list`.

I now get the following error, which might be related:
```python
Traceback (most recent call last):
  File "test.py", line 2, in <module>
    import skywriter
  File "/usr/local/lib/python2.7/dist-packages/skywriter-0.0.2-py2.7.egg/skywriter.py", line 31, in <module>
    i2c = SMBus(i2c_bus_id())
IOError: [Errno 2] No such file or directory
```

Gadgetoid, 2015-01-26:
This would be because you haven't explicitly enabled i2c, which is required for Skywriter. Try our easy-way:

    curl get.pimoroni.com/i2c | bash
    sudo reboot

Follow the steps presented from the first command, reboot, and your i2c should be set up.

goliatone, 2015-01-26:
@Gadgetoid thanks for the reply. Will check it out once I get back home from work. Was all this supposed to work out of the box, am I missing a 101 article I did I not read or something?

goliatone, 2015-01-26:
I installed `i2c` successfully but the error persists.

Gadgetoid, 2015-01-26:
Have you run "rpi-update" recently? If so, read: http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314

Gadgetoid, 2015-01-26:
Fixed this with better documentation for now: 8f110a918353b8cc00858c6e5011d9e13afc8d3f

But I believe the above forum thread, and ongoing problems with the new device-tree kernel represent a continuing issue.

goliatone, 2015-01-26:
Indeed, I had run `rpi-update` recently. The link provided a solution and **skywriter-hat** is now working.
Just as a reference. I had to add the option `device_tree=` to config.txt and reboot

```terminal
sudo nano /boot/config.txt
#Add option: device_tree=
sudo shutdown -r now
```

closed 2015-01-26 by goliatone