pimoroni/skywriter-hat #3: Installing bash script fails

goliatone, 2015-01-26:
Running the command, execute but fails because `smbus` dependency:
```terminal
curl -sSL get.pimoroni.com/skywriter | bash
```

The raspberry pi has python versions 2.6 and 2.7 with 2.7 being the active one.

Relevant output:
```teminal
Python 2 module found. Reinstall/Update? [y/N] y
Collecting skywriter
  Using cached skywriter-0.0.3.tar.gz
Collecting smbus (from skywriter)
  Could not find any downloads that satisfy the requirement smbus (from skywriter)
  No distributions at all found for smbus (from skywriter)
Python 2 library install failed!
```

Full output:
```terminal
This script will install Skywriter libraries
and dependencies for both Python 2.

This process may take several minutes.

--- Warning ---

Always be careful when running scripts and commands
copied from the internet. Ensure they are from a
trusted source.

If you want to see what this script does before
running it, you should run:
    \curl -sS get.pimoroni.com/skywriter

Do you wish to continue? [y/N] y

Skywriter requires I2C, enable now? [y/N] y

I2C Already Enabled

Checking install requirements...

dpkg-query: no packages found matching python-pip
Updating package indexes...
Get:1 http://archive.mene.za.net wheezy Release.gpg [198 B]
Get:2 http://mirrordirector.raspbian.org wheezy Release.gpg [490 B]                                                       
Get:3 http://raspberrypi.collabora.com wheezy Release.gpg [836 B]                                                         
Get:4 http://vontaene.de . Release.gpg [198 B]                                                                            
Get:5 http://mirrordirector.raspbian.org wheezy Release [14.4 kB]                                                         
Hit http://vontaene.de . Release                                                  
Get:6 http://raspberrypi.collabora.com wheezy Release [7,511 B]                   
Hit http://archive.raspberrypi.org wheezy Release.gpg                             
Ign http://vontaene.de . Release                                                                                          
Get:7 http://raspberrypi.collabora.com wheezy/rpi armhf Packages [2,214 B]                                                
Get:8 http://mirrordirector.raspbian.org wheezy/main armhf Packages [6,894 kB]                                            
Hit http://archive.raspberrypi.org wheezy Release                                                                         
Get:9 http://archive.mene.za.net wheezy Release [4,463 B]                                                                 
Get:10 http://archive.mene.za.net wheezy/contrib armhf Packages [11.0 kB]                                                 
Hit http://archive.raspberrypi.org wheezy/main armhf Packages                                                             
Hit http://vontaene.de ./main armhf Packages                                                                              
Ign http://vontaene.de ./main Translation-en_GB                                                                           
Ign http://vontaene.de ./main Translation-en                                                                              
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en_GB                                                         
Ign http://raspberrypi.collabora.com wheezy/rpi Translation-en                                                            
Ign http://archive.mene.za.net wheezy/contrib Translation-en_GB                
Ign http://archive.mene.za.net wheezy/contrib Translation-en                   
Ign http://archive.raspberrypi.org wheezy/main Translation-en_GB                                                          
Ign http://archive.raspberrypi.org wheezy/main Translation-en                                                             
Hit http://mirrordirector.raspbian.org wheezy/contrib armhf Packages                                                      
Hit http://mirrordirector.raspbian.org wheezy/non-free armhf Packages                                                     
Hit http://mirrordirector.raspbian.org wheezy/rpi armhf Packages                                                          
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en_GB                                                   
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en                                                      
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en_GB                                                      
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en                                                         
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en_GB                                                  
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en                                                     
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en_GB                                                       
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en                                                          
Fetched 6,935 kB in 2min 29s (46.3 kB/s)                                                                                  
Reading package lists... Done
W: GPG error: http://vontaene.de . Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY F0DAA5410C667A3E

Installing Skywriter requirements...
dpkg-query: no packages found matching python-pip

Checking for python-pip...
python-pip required. Installing.
Running sudo apt-get install python-pip

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following extra packages will be installed:
  python2.6 python2.6-minimal
Suggested packages:
  python2.6-doc binfmt-support
Recommended packages:
  python-dev-all
The following NEW packages will be installed:
  python-pip python2.6 python2.6-minimal
0 upgraded, 3 newly installed, 0 to remove and 24 not upgraded.
Need to get 3,962 kB of archives.
After this operation, 13.2 MB of additional disk space will be used.
Get:1 http://mirrordirector.raspbian.org/raspbian/ wheezy/main python2.6-minimal armhf 2.6.8-1.1 [1,409 kB]
Get:2 http://mirrordirector.raspbian.org/raspbian/ wheezy/main python2.6 armhf 2.6.8-1.1 [2,442 kB]                       
Get:3 http://mirrordirector.raspbian.org/raspbian/ wheezy/main python-pip all 1.1-3 [112 kB]                              
Fetched 3,962 kB in 38s (103 kB/s)                                                                                        
Selecting previously unselected package python2.6-minimal.
(Reading database ... 78209 files and directories currently installed.)
Unpacking python2.6-minimal (from .../python2.6-minimal_2.6.8-1.1_armhf.deb) ...
Selecting previously unselected package python2.6.
Unpacking python2.6 (from .../python2.6_2.6.8-1.1_armhf.deb) ...
Selecting previously unselected package python-pip.
Unpacking python-pip (from .../python-pip_1.1-3_all.deb) ...
Processing triggers for man-db ...
Processing triggers for desktop-file-utils ...
Processing triggers for menu ...
Setting up python2.6-minimal (2.6.8-1.1) ...
Linking and byte-compiling packages for runtime python2.6...
Setting up python2.6 (2.6.8-1.1) ...
Setting up python-pip (1.1-3) ...
Processing triggers for python-support ...
Processing triggers for menu ...

Checking for python-dev...
python-dev already installed.

Checking for Python 2 library...

Found!

Python 2 module found. Reinstall/Update? [y/N] y
Collecting skywriter
  Using cached skywriter-0.0.3.tar.gz
Collecting smbus (from skywriter)
  Could not find any downloads that satisfy the requirement smbus (from skywriter)
  No distributions at all found for smbus (from skywriter)
Python 2 library install failed!
```

Gadgetoid, 2015-01-26:
Good spot. Looks like it's not installing python-smbus as a requirement. 

I'll get this sorted ASAP, but for future reference this issue should probably be filed against the install script repository, which is here: https://github.com/pimoroni/get

Thanks for your extremely concise information, though, it's a blessing to get an issue that's more than "I have a problem" :+1:

goliatone, 2015-01-26:
I was unaware of the **pimoroni/get** repo. Will file issues against as needed- hopefully not needed :smile: .

Gadgetoid, 2015-01-26:
Fixed in: https://github.com/pimoroni/get/commit/1a7c841cc985cc583e5a4ca8b26de2ba136d9e1a

I have synced this to get.pimoroni.com, if you're in a position to retry the installer I'd be most thankful for your feedback!

goliatone, 2015-01-26:
seems to be working

closed 2015-01-26 by goliatone