pimoroni/skywriter-hat #4: IOError: [Errno 2] No such file or directory - after running rpi-update

Gadgetoid, 2015-01-26:
We've become aware of an issue with the new device-tree enabled kernel which requires i2c to be enabled in config.txt ( although I've been told this behaviour is being reverted so that i2c is enabled, but modules not inserted by default ).

If you're a regular uses of rpi-update, and you're seeing an error like this:

```python
Traceback (most recent call last):
  File "test.py", line 2, in <module>
    import skywriter
  File "/usr/local/lib/python2.7/dist-packages/skywriter-0.0.2-py2.7.egg/skywriter.py", line 31, in <module>
    i2c = SMBus(i2c_bus_id())
IOError: [Errno 2] No such file or directory
```

Then please read this thread: http://www.raspberrypi.org/forums/viewtopic.php?f=28&t=97314

If you're here because you're seeing a similar error, but don't use rpi-update, then please try our installer script:

```bash
curl get.pimoroni.com/skywriter | bash
```

closed 2015-03-24 by Gadgetoid

CoffeeAndCream, 2016-01-28:
Hey guys, thanks this has been a life saver for me.

UnJavaScripter, 2019-02-17:
Leaving this here in case someone else gets to this issue looking for a solution for the same problem but with the rainbowhat, like I did:

Can confirm this also works for the Rainbow HAT, just replace the route in the URL:
`curl get.pimoroni.com/rainbowhat | bash`