pimoroni/skywriter-hat #7: examples/test.py just prints firmware info

besherman, 2015-07-09:
I've enabled i2c and installed the library, but when I try to run the example program all that happens is that it prints the firmware info, over and over again until i ctrl-c out of it. 

If I move my hand over or tap the sensor nothing happens, except the same stream of firmware info.

> Got firmware info
> 1.1.18;p:Hills
> Got firmware info
> 1.1.18;p:Hills
> Got firmware info
> 1.1.18;p:Hills

besherman, 2015-07-09:
I took a peek in `handle_firmware_info()` and the value of `d_fw_valid` is 0x0A which according to the data sheet means "InvalidFW: An invalid GesticIC Library was stored, or the last update failed". Unfortunately this means nothing to me.

Gadgetoid, 2015-07-09:
Looks like the firmware on the HAT is borked- if you want, I'll guide you through the process of re-flashing it, which may get you back up and running. If all else fails we'll replace it. I'll dig up the re flash steps ASAP- if you're up for giving it a try, drop me an email to phil at Pimoroni.com.

closed 2015-09-07 by Gadgetoid

Enkerli, 2016-05-17:
Is that stream of firmware info a sufficient sign that the firmware might be borked?
````
Got firmware info
1.1.18;p:Hills
````

Kclemen, 2016-12-09:
Hi, I'm having the same issue. What was the solution??