pimoroni/skywriter-hat #9: Reset not always HIGH?

paradajz, 2015-10-19:
Please enable internal pull-up for reset pin on Arduino when Skywriter is being initialized. Sometimes on start-up, sensor is "frozen", that is, it doesn't work until next reset. I've found out that enabling internal pull-up resistor on Arduino solves the issue, but I'm not sure why.

Change line 14 in skywriter.cpp from:
pinMode(this->rst, INPUT);
to:
pinMode(this->rst, INPUT_PULLUP);

Gadgetoid, 2015-10-19:
Looking back at the logic surrounding reset on Skywriter, it's sufficiently weird enough to make me wonder what on earth I was thinking at the time.

Why would reset ever need to be an input at all ( https://github.com/pimoroni/skywriter-hat/blob/master/arduino/Skywriter/skywriter.cpp#L11-L13 )? I suspect it should remain an output, but instead set to `HIGH`. Thanks for pointing it out, I'll look into it!

Gadgetoid, 2015-10-23:
This is, hopefully, fixed in 81e184edb4ff7d8dbf9a0755c3415d727c931a32

paradajz, 2015-10-23:
It won't work at all now actually.

Gadgetoid, 2015-10-23:
That's especially odd- any idea why? Famous last words; In both cases it works for me.

paradajz, 2015-10-27:
This is becoming annoyingly weird. At first it didn't work at all, then I tried to upload the code couple more times and it worked. Then I tried to reset the Arduino few more times and it stopped working again. Ugh. So, again, sometimes it works, sometimes it doesn't. I'm still trying to make sense from sensor datasheet and reset logic. Have you tried to reset your Arduino few times in a row to see does it work afterwards?

Gadgetoid, 2015-10-27:
I'm using an Arduino Uno. It sounds like there could be an electrical problem with your Skywriter- I'm guessing all the wiring will have been eliminated by trying it with multiple different boards. As frustrating as that is, I can't think of another reasonable explanation.

paradajz, 2015-10-27:
That's possible as well. I'm going to try it on another sensor today to see how it behaves.

closed 2016-03-14 by Gadgetoid