pimoroni/skywriter-hat #15: Autopy not installed

sandyjmacdonald, 2016-05-18:
With the get installer, it looks like autopy, required for the keyboard and mouse examples, isn't installed. Further, trying to pip install autopy seems to fail also on the most recent version of Jessie.

sandyjmacdonald, 2016-05-18:
Git clone and install of autopy doesn't seem to install properly either.

Fails with an `X11/Xlib.h: No such file or directory` error.

Gadgetoid, 2016-05-18:
This will be caused by the lack of `libx11-dev` which might have been previously installed by default, or perhaps never installed at all under Raspbian. In either case, I think it's a job for documentation and a tweak to the installer scripts.

Summoning @RogueM since for some reason something about this rings a bell.

RogueM, 2016-05-18:
I don't recall the details, but yes, it was excluded for that reason or something similar.

Anything that is not critical for the lib itself to function is collated in the moreaptdep/morepipdep array and the user has the choice to install them or not... I include as many as practical to support the examples in the repo, but they have to lend themselves to not trip the installer and leave the user wondering whether the (library) install was succesful.

... so, yeah, the o-l-i can be modified, but there would have to be a chance of success under normal conditions. The 'more' routines also currently do not support separate handling for Jessie vs Wheezy, although it could be done without too much troubles, if say it was possible to get autopy working on Jessie but not Wheezy (though I don't think that's the case).

closed 2016-07-06 by Gadgetoid