pimoroni/skywriter-hat #21: Exception("Skywriter encoutered nore than 10 consecutive I2C IO errors!")

alesms, 2017-03-10:
Hi,
i've installed the new skywriter 3D and all works fine, but almost one or two times per day i've got this error and the script doesn't work anymore and the only way is to restart the script.
The error that appears is:

Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/skywriter.py", line 107, in run
    if self.todo() == False:
  File "/usr/local/lib/python2.7/dist-packages/skywriter.py", line 313, in _do_poll
    raise Exception("Skywriter encoutered nore than 10 consecutive I2C IO errors!")

and the script is:

import skywriter
import signal

@skywriter.flick()
def flick(start,finish):
  try:
    print('Got a flick!', start, finish)
    if start == "west":
      print ("got west")
      
  except:
    print "ERROR"

signal.pause()

Gadgetoid, 2017-03-13:
This is curious. Is this using a Skywriter board, rather than a HAT? How long is your wiring between it and your Pi?

In practise IO errors should occur infrequently unless there's a wiring problem causing crosstalk or picking up environmental noise. That would be my first port of call for diagnosing this problem.

alesms, 2017-03-13:
hi, the wiring is done by a default cable that you can buy to wiring some sensor to raspberry (i think 20cm approximately).
 Yes i use a SkyWriter board; 
this https://shop.pimoroni.com/products/skywriter

but, anyway, is there a way to jump this error and reset the ambient to work again without restart the script?
beacause in this moment, when the skywriter.py script goes in exception, the script crashes and no longer able to receive other commands.
Thank you

Gadgetoid, 2017-03-14:
You could remove these two lines from the library: https://github.com/pimoroni/skywriter-hat/blob/master/library/skywriter.py#L284-L285

alesms, 2017-03-15:
i tried but the result is the same....you don't have the exception but the script doesn't work anymore

alesms, 2017-03-16:
I thought that on my PI I'm using an LCD screen 20 * 4, which also uses the I2C protocol.
It could be the problem?

Gadgetoid, 2017-03-17:
That could very easily be the problem- are you using a separate Python script to communicate with the LCD?

alesms, 2017-03-22:
no the script is the same. But, maybe, i found the error!
Under the skywriter i have an RF transmitter and I noticed that, almost everytime, when I send an RF command, I get the error in the skywriter script. 
So I covered the below part of the skywriter with aluminum foil and is not giving the problem for some days.

closed 2017-04-12 by Gadgetoid