pimoroni/skywriter-hat #27: Skywriter hat cannot pick up on flicks - broken/not working?

nickdebCompApps, 2018-01-18:
I have recently purchased a skywriter hat and have followed all directions on installing scripts and inserting hat into GPIO pins. When I run test.py, the script does not pickup on flicks, and once, say, I touch the board, the touch is registered but then it continues to print touch even when I am not touching it. When I press control + C to cancel the script running, it continues to print out test.py. I am wondering if this is a programming error or something on the hardware side.

Gadgetoid, 2018-03-23:
Sorry for the delay responding- did you find a solution to this problem or manage to obtain a replacement from us?

UnsignedArduino, 2019-12-21:
You have to touch a edge pad that does not have a grid on it and flick to the opposite pad that does not have a grid on ot

closed 2023-01-20 by Gadgetoid