pimoroni/skywriter-hat #28: SW_CIRCLE_C(C)LOCKWISE gesture

Mynasru, 2018-03-24:
I cannot get `SW_CIRCLE_CLOCKWISE` and `SW_CIRCLE_CCLOCKWISE` working, when are these supposed to be activated (what gesture) and are these triggered by `void gesture(unsigned char type)` or by another function?

Also the documentation on the Arduino library and examples is very scarce, it could use an update.

Gadgetoid, 2018-03-27:
You're right- looks like I've neglected the Arduino library.

If I remember correctly it's the "onAirwheel" handler that you need.

It's a tricky gesture to get right, but basically involves spinning your finger around above the Skywriter in a clockwise or counter-clockwise direction.

balamsoto, 2018-05-03:
having this into arduino / teensy arch can lead to more sales
Please update the lib and enable all the magic of this hardware

closed 2023-01-20 by Gadgetoid