pimoroni/skywriter-hat #32: worker.stop() AttributeError: 'NoneType' object has no attribute 'stop'

bebon901, 2019-02-25:
I get this error after running test.py (sudo python test.py):
`Traceback (most recent call last):
  File "umouse.py", line 2, in <module>
    import skywriter
  File "/usr/lib/python2.7/dist-packages/skywriter.py", line 502, in <module>
    i2c.write_i2c_block_data(SW_ADDR, 0xa1, [0b00000000, 0b00011111, 0b00000000, 0b00011111])
IOError: [Errno 121] Remote I/O error
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/skywriter.py", line 493, in _exit
    stop_poll()
TypeError: 'NoneType' object is not callable
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/skywriter.py", line 493, in _exit
    stop_poll()
TypeError: 'NoneType' object is not callable
`
Can anyone help me resolve this issue?

i2c detect -y 2 gives me:
`     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- 3a -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
`
and the others all produce a blank report.
Thank You.

Gadgetoid, 2019-02-25:
Judging from the i2c-detect output, the HAT isn't being detected by your Pi at all. How is it connected?

Toby56, 2020-01-28:
My Skywriter hat broke suddenly and now it just doesn't work like this. Do you know if there is anything I can do?

UnsignedArduino, 2020-01-28:
Does it look exactly the same like in the pictures, or did you like, drop it and the chip broke off?

Toby56, 2020-01-28:
It does look like like the pictures, it's not a physical problem. I think I must have connected it wrong briefly or maybe because I plugged it in while the pi was turned on, so it's probably my fault. It broke before I really got to do anything with it, so I'm sad.

UnsignedArduino, 2020-01-28:
Do you have any other devices using I2C that still work on your Raspberry Pi? 
Maybe I2C is off, or maybe if you soldered the Raspberry Pi headers yourself, it might be loose.

Toby56, 2020-01-28:
I haven't checked with any other devices, but it's highly unlikely. The headers are fine and I definitely made sure I2C was still turned on. Thanks for the help : )

UnsignedArduino, 2020-01-28:
Your welcome! 
Sorry I couldn't help any more, maybe someone else with more experience could help. 
Maybe you could return it? Or ask in the Pimoroni forums or contact Pimoroni support.

Toby56, 2020-01-28:
Yeah thanks. I didn't buy it directly from Pimoroni, so I don't know if I can return it. Maybe I'll try Pimoroni support!

UnsignedArduino, 2020-01-28:
Hope you Skywriter works!

Gadgetoid, 2020-03-16:
@Toby56 did you have any luck with Pimoroni support? Closing this now because I don't think there's anything I can do to help, but feel free to re-open if neccessary.

closed 2020-03-16 by Gadgetoid