pimoroni/skywriter-hat #34: Skywriter XL XYZ positions inaccurate

NickHarmer, 2020-09-05:
I have an issue with my SkyWriter XL,  using the latest RPi library and having updated to the latest firmware, I am finding that the XYZ coordinates returned by the move reports are highly inaccurate and inconsistent.

For example:

* in the Horizontal (X) axis, the values I get back range from about 0.25 at the extreme left hand side to 1.0, but it reaches 1.0 when my finger is barely half way across the sensor. 
* in the vertical (Y) axis, it stays at 0.0 until my finger is about a quarter of the way up from the bottom and never goes above 0.75
* the Z axis, by contrast, seems to be about correct 

Is this behaviour normal for the XL, or is there a problem with my device?

NickHarmer, 2020-09-05:
Ignore me - environmental problem!

Gadgetoid, 2020-09-08:
Which process did you use to update the firmware? It's possible you could have Skywriter HAT firmware instead of XL.

I'll be honest- it's a *long* *LONG* time since I've touched any of the code relating to these boards, so I'm at risk of being about as useful as a chocolate teapot here but I'll try and dredge up the relevant memories.

Gadgetoid, 2020-09-08:
Some steps for the XL Skywriter firmware update procedure:

First, get the firmware:

```
wget http://get.pimoroni.com/skywriter-fw.tgz
```

Extract:

```
tar -xzf skywriter-fw.tgz
```

Then run it:

```
cd skywriter-fw
sudo python3 update.py
```

Gadgetoid, 2020-10-28:
Did the above help at all?